hiveeyes/hanimandl #3: Poti does not work correct randomly.

ClemensGruber, 2019-02-17:
The potentiometer should e.g. adjust the opening angle or act as menu item selector. It is not working randomly an my hardware setup. Don't know if this is a hardware issue or a software bug. 

What I did: 
- used an other poti 
- used an other pin 34 instead of pin 39 because pin 39 has some capacitors build in 
but without any change.

ClemensGruber, 2019-02-17:
Reason was a broken hardware part. I did test the doubtful poti vs. the second one by multimeter only and the measurement seemed to be similar, but I did not test both side by side in the circuit. After replacing the first one by a new one things working smoothly!

closed 2019-02-17 by ClemensGruber

ImkereiWetzel, 2019-02-17:
Just a comment from me - I use a 20k poti and it also works really smooth (maybe even smoother, as i dont see the "flickering" between two menu-items, as shown in the video on facebook