hiveeyes/hanimandl #5: Weight cell readings show crazy values (-2147..g)

ImkereiWetzel, 2019-02-18:
You get crazy reading, no change in weight and you had checked your pinning twice?
It might be related to an internal overflow - please test if a calibration AND a tara procedure (see -> setup) helps

ClemensGruber, 2019-02-18:
We shoulde force the user -- or at least give a hint -- to calibrate and tare the scale in case there was no calibration done before. We can check this by having a look at the variables used via Preferences.h / "EEPROM" in the old Arduino universe.

ClemensGruber, 2020-06-18:
There is a hint now to calibrate the scale in case it is not done yet, see https://github.com/ClemensGruber/hani-mandl/issues/2

closed 2020-06-18 by ClemensGruber

mahrmediait, 2021-02-22:
Hallo, leider habe ich beim Tare immer den Wert 21474. habe es mit unterschiedlichen Wägezellen und verschiedenen HX711 versucht. Bei einem simplen Test Programm klappt alles.

jemand eine Idee?

MKO1640, 2021-05-04:
Moin, Du hast vor dem Tara sicherlich kalibriert. Oder? Eine schlechte Verbindung sollte es auch nicht sein, wenn Du schon verschiedene Wägezellen und HX711 probiert hast. So Tippe ich mal, auf die eine Falsche HX711 Library. Hast du die Vorgeschlagene genommen? Bist Du mit VSCode +PlatformIO oder mit der Arduino IDE unterwegs?

EVTl. ist aber auch die Pinconfig vom HX711 nicht OK, welche Meldung macht er denn beim Hochfahren?