hiveeyes/hanimandl #6: Crash

Devil-of-Chaos, 2020-05-24:
Hab es zwar noch nicht debugged und wollte es mir noch anschauen und dann nen Push schicken. Aber im wifi branch gibt es einen Absturz wenn man in der Automatik bei der Anzeige start das Glas wieder entfernt. Evtl. kann dies noch jmd verifizieren.

ClemensGruber, 2020-05-25:
Hi, welchen wifi-branch, sehe hier im Repo keinen? Ist das vielleicht dein lokaler? WiFi brauchen wir aktuell für den HaniMandl nicht.

ClemensGruber, 2020-05-25:
So gerade mal mit dem aktuellen code in _master_ getestet. Wenn ich das Glas auf die Waage stelle erscheint "Start" für ca. 3 Sekunden, dann startet der Servo und öffnet den Hahn, nun wird das aktuelle Gewicht angezeigt, und nach einer Zeit der Hahn beim erreichen des Sollgewichts geschlossen. Alles normaler Betrieb. 

Nun stelle ich das Glas auf die Waage, "Start" erscheint und ich entferne das Glas nach ca. einer Sekunde, wenn also noch "Start" im Display steht. Bei mir öffnet der Hahn kurz und schließt dann! Das Verhalten ist nicht optimal, weil ja in der viertel Sekunde Honig rausfließen kann, crashen tut aber nix.

closed 2020-05-25 by ClemensGruber

Devil-of-Chaos, 2020-05-25:
Hi sorry meine diesen Branch "a-holzhammer_new-wifi-kit-32" 
Ja ein Crash direkt ist es nicht. Genau arm geht kurz auf und wieder zu. Bei dem holzhammer branch taucht kurz im Bildschirm der Text Boot auf. Deshalb Crash.