hiveeyes/hanimandl #14: Using PlatformIO on Windows natively

amotl, 2020-11-19:
Dear Clemens,

#8 adds a wash & go build command for *nix environments, it should work out of the box for Linux and macOS. However, I would be interested if we can make this also work on Windows without any significant efforts.

So, by following [1,2,3], the setup might be as easy as:

### Install prerequisites
```
# 1. Install Chocolatey
@"%SystemRoot%\System32\WindowsPowerShell\v1.0\powershell.exe" -NoProfile -InputFormat None -ExecutionPolicy Bypass -Command " [System.Net.ServicePointManager]::SecurityProtocol = 3072; iex ((New-Object System.Net.WebClient).DownloadString('https://chocolatey.org/install.ps1'))" && SET "PATH=%PATH%;%ALLUSERSPROFILE%\chocolatey\bin"

# 2. Install Python
choco install python --version=3.8.6

# 3. Install PlatformIO
pip install platformio
```

### Acquire sources
- Download and extract sources from https://github.com/ClemensGruber/hani-mandl/archive/master.zip.
- Change to directory using `cd hani-mandl-master`.

### Build firmware
```
platformio run
```

With kind regards,
Andreas.

[1] https://chocolatey.org/docs/installation
[2] https://chocolatey.org/packages/python/3.8.6
[3] https://docs.platformio.org/en/latest/core/installation.html#python-package-manager

MKO1640, 2021-05-14:
Moin Andreas, Ja es funktioniert. Ich habe bei mir zwar schritt 1 und 2 ausgelassen, da Python bei mir schon installiert ist, aber `pip install platformio` und `platformio run` funktioniert bei mir in der Powershell danach.  
Bin mir aber nicht sicher ob bei Windows zusätzlich noch ein C/C++ Compiler installiert werden muss. Habe jetzt auf die schnelle nicht rausfinden können welchen er nativ verwendet.

MKO1640, 2021-05-18:
sehr schön sogar ein in der platformio.ini vor-konfigurierte [env] mit Buildflags und spezifischen Library´s funktioniert. 
`pio run -e DevKitC_240_320_SPI_TFT -t upload`
die platformio.ini dafür habe ich aus meinem [Dev-Micha](https://github.com/MKO1640/hani-mandl/blob/dev-Micha/platformio.ini) Branch genommen