hiveeyes/hanimandl #18: Adding unit tests

amotl, 2021-01-08:
Hi there,

I just discovered three nice projects and wanted to share them with you. They are AUnit [1], AUniter [2] and UnixHostDuino [3] and they will compile and execute unit tests for Arduino both natively on Linux or MacOS and by uploading to boards.

With kind regards,
Andreas.

[1] https://github.com/bxparks/AUnit
[2] https://github.com/bxparks/AUniter
[3] https://github.com/bxparks/UnixHostDuino

amotl, 2021-01-15:
@SBajonczak also referenced [1] at [2].

[1] https://docs.platformio.org/en/latest/plus/unit-testing.html
[2] https://github.com/SBajonczak/hivemonitor/tree/master/test

amotl, 2024-04-13:
We just discovered how Rob Tillaart is testing his HX711 library on CI/GHA. He is using [ArduinoCI](https://github.com/Arduino-CI/arduino_ci), and it looks like it works pretty well **enabling running unit tests without hardware present**?

### References
- https://github.com/RobTillaart/HX711/tree/master/test
- [.github/workflows/arduino_test_runner.yml](https://github.com/RobTillaart/HX711/blob/9eebc4bb8662032f11aa02ef2d4adcbaaa8e796f/.github/workflows/arduino_test_runner.yml#L10-L17)

RobTillaart, 2024-04-14:
> We just discovered how @RobTillaart is testing his HX711 library on CI/GHA. He is using [ArduinoCI](https://github.com/Arduino-CI/arduino_ci), and it looks like it works pretty well **enabling running unit tests without hardware present**. Thanks, Rob!
> 
> ### References
> * https://github.com/RobTillaart/HX711/tree/master/test
> * [.github/workflows/arduino_test_runner.yml](https://github.com/RobTillaart/HX711/blob/9eebc4bb8662032f11aa02ef2d4adcbaaa8e796f/.github/workflows/arduino_test_runner.yml#L10-L17)

Did a quick test, cloned this repo and added the Arduino unit test (quick and dirty copy / strip) to it.
- Arduino-CI
- lint
- json check
The latter two do nothing so may be removed.

It appeared that the PlatformIO workflow tries to run the code in the test folder where the unit tests reside and fails.
See - https://github.com/RobTillaart/hanimandl/actions

Note that I use the Arduino-CI mainly for libraries, and I used it for an application by adding an "examples" folder, in which the applications are copied as if they were examples from a library.
See - https://github.com/RobTillaart/MultiSpeedI2CScanner

Configured for ESP32 platforms in .arduino-ci.yml  (top folder repo)  + 3 needed libraries.
as hanimandl.ino uses ESP32Servo it can only compile on ESP32.

Added a badge for the Arduino CI in my repo (readme,md) for test. 
You need to patch the URL in the badge of course.
[image]

Hope this helps you to get started.
Rob

RobTillaart, 2024-04-14:
If you want I can make a PR of my cloned library.

amotl, 2024-06-23:
Hi Rob.

I missed to respond here properly. Thank you so much for looking into this. If you could submit your improvements to the CI configuration as a PR, that would be nice indeed!

With kind regards,
Andreas.

RobTillaart, 2024-06-23:
@amotl 

Created - https://github.com/hiveeyes/hanimandl/pull/62

amotl, 2024-06-23:
Thanks a stack. Approved CI runs, and they succeeded on the first attempt. Excellent!