hiveeyes/hanimandl #23: Add support for TTGO T-Display

amotl, 2021-02-10:
Hi there,

based on #21, we might want to add first-citizen support for the TTGO T-Display, see also [1] and [2]. I will try to get some slot for that together with @ClemensGruber. Thanks already for outlining the nitty gritty details to be considered on the implementation side within [2].

With kind regards,
Andreas.

[1] https://community.hiveeyes.org/t/ttgo-t-display-als-alternative-zum-heltec/3733
[2] https://community.hiveeyes.org/t/prototypische-verwendung-des-ttgo-t-display-als-hanimandl-board/3830

andreash-esp, 2021-02-24:
My personal review:
https://hanimandl.de/2021/02/23/ttgo-t-display-hanimandl/
A great device, but not fit for this purpose

MKO1640, 2021-04-19:
I think it is sufficient for this purpose.
Bringing some color is something beautiful.
And the missing pins are not needed at all.
In my fork, I'm a little further.
Also with regard to #21 .

But I can not test the downward compatibility with the Heltec because I do not have one at the moment.

amotl, 2023-01-31:
Hi,

I just discovered that the link shared by @MKO1640 is pointing to the wrong destination. The correct one is https://github.com/MKO1640/hani-mandl/commits/dev-Micha.

With kind regards,
Andreas.