hiveeyes/hanimandl #28: Servohalter/Heilbronn ist krumm. Soll das so sein? :)

freaky-m0, 2021-04-23:
Hi,

ist mir nur aufgefallen, weil ich es zu dieser Seite auf dem Druckbett ausgerichtet hatte. Ist irgendwie krumm der Servohalter von der Heilbronn Variante. Soll das so sein? Ist das erste Mal, dass ich den Bausatz drucke. (Die Pfeile sind zur Verdeutlichung)

[image]

Danke & Gruß

MKO1640, 2021-05-05:
Sollte eigentlich nicht so sein, es macht schon Sinn, ihn in dieser Ausrichtung zu drucken. Bei nicht perfekter Layerhaftung würde er sonnst leichter brechen. 
Da ich ihn nicht konstruiert habe kann ich leider nichts zu den Beweggründen sagen. Es ist ja etwas aufwendiger den Halter mit der Schrägung zu Konstruieren, denke es gibt da schon einen Grund.

amotl, 2024-01-22:
Hallo @freaky-m0,

wahrscheinlich werden wir nicht mehr herausfinden, warum das Bauelement so konstruiert wurde, aber ggf. kann der ursprüngliche Autor hier noch antworten. Einstweilen schließe ich das Ticket, but feel free to re-open if applicable.

Viele Grüße,
Andreas.

closed 2024-01-22 by amotl