hiveeyes/hanimandl #30: Does hani-mandl run on Arduino?

ganafets82, 2021-11-28:
Hi,
Could I run it on Arduino, or its the ESP32 choosen on purpose?

Thanks for this project!

Stefan

ClemensGruber, 2022-01-21:
The reason for choosing the Heltec board was the onboard display and the cheap price. :-) There was an ESP32 microcontroller on board but from a technical side a M0 would also be sufficient. BUT with the ESP32 as given we have choosen the appropriate libs for the ESP and not for Arduino. E.g. we are using the ESP32Servo lib. So in case you want to try the code with an Arduino you have to replace at least this lib. But we do not use special features of the ESP or onboard WiFi. So it should work on an Arduino also with slight adjustments but not as is in the repo.

amotl, 2024-01-22:
Dear Stefan,

thank you for writing in. Your [dispenser](https://github.com/ganafets82/dispenser) project looks like it is pretty similar to hanimandl, right?

Through `#include <Servo.h>`, we can see you are using the [vanilla Arduino Servo library](https://www.arduino.cc/reference/en/libraries/servo/). However, this library is not suitable for Espressif chips. A library with apparently the same interface, but suitable for ESP32, would be [ServoESP32](https://github.com/RoboticsBrno/ServoESP32) ([ServoESP32 on PlatformIO](https://registry.platformio.org/libraries/roboticsbrno/ServoESP32)).

However, I also recognize that [ESP32Servo](https://registry.platformio.org/libraries/madhephaestus/ESP32Servo) is doing quite the same, so it may be possible to swap it to use the vanilla Arduino library, by just using a different `#include`?

With kind regards,
Andreas.

ganafets82, 2024-01-23:
Hi Andreas,
The dispenser is somehow similar but not as advanced as the hanimandl. I used this several weeks a year now for filling hot juice. It just works will since the beginning, there was no need to change something except the red part (visible at thingiverse) is aluminium now.
It has no display and I will not develope this further and I plan to use the hanimandl code or instead.

I cant tell you about the servo library on ESP, I never used it.
I would wonder if it would not work similar.

Best wishes
Stefan

ClemensGruber, 2024-01-23:
Bit OT, but I have to addmit: A very nice idea to use a tube and this nice eccentric mechanism to press of a hose for filling up liquids! 

https://www.thingiverse.com/thing:4631906

I spent this nice idea a posting under: https://community.hiveeyes.org/t/fur-flussigkeiten-abgequetschter-schlauch-statt-quetschhahn/5202 feel free to jump in, Stefan / @ganafets82 ! And many thanks to Andreas / @amotl as "Trüffelschwein"! :-)