hiveeyes/hanimandl #31: Missing License

fubian, 2021-12-05:
This project should include a license such like the MIT or the GPLv3 license to protect your work appropriately.

amotl, 2023-01-16:
Dear Fabian,

I absolutely agree. I propose to use the GPL-3.0 license, to align this with the other Arduino code on the Hiveeyes repository [^1]. On this matter, I hope the original authors agree.

Please let us know if you think differently, @MarcWetzel, @andreash-esp, and @ClemensGruber. If anyone of you could reach out to Marc Vasterling, Marc Junker, Johannes Kuder, and Jeremias Bruker, that would be nice. Otherwise, I would assume that everyone will agree on changing the license to GPLv3 without further ado, in order to protect the work appropriately.

In case there are any concerns: Any other non-FOSS forks will not be impacted, the license change will probably only apply from the date of issuance.

With kind regards,
Andreas.

[^1]: https://github.com/hiveeyes/arduino

amotl, 2024-01-23:
Hi again, and a happy new year. We've implemented your suggestion on behalf of GH-45.

closed 2024-01-23 by amotl