hiveeyes/hanimandl #32: Random crashes when using the rotary encoder

johannesarndt, 2022-08-09:
Compiled the code and flashed it on the Heltec. The program starts and when I try to change settings the response is very slow and by using the rotary encoder the program crashes at a random point. 
The serial monitor shows some kind of exception in the task scheduler.
Going to investigate the reasons and post the error message later.

johannesarndt, 2022-08-10:
assert failed: taskSelectHighestPriorityTaskSMP tasks.c:3446 (xTaskScheduled == pdTRUE)

Backtrace:0x40084688:0x3ffbd4000x3ffbd4bd:0x3ffbd4e0  |<-CORRUPTED

ELF file SHA256: 0000000000000000

doc2005, 2022-11-13:
same issue on esp32 wroom.
setting version of espressif to 3.5.0 does not help at all

amotl, 2023-06-22:
Dear @InstructionDecode,

thank you for submitting GH-33, and good to hear that this works for you. @doc2005: Could you elaborate what does not work?

Maybe @ClemensGruber outlined the correct solution how to modify the code at https://community.hiveeyes.org/t/kann-ich-auch-das-heltec-wifi-kit-32-in-der-neuen-version-v3-verwenden/4696/37?

> For me the code runs stable if I comment out the line about 307 (depending on the code version):
> // if ( rotating ) delay (1); // wait a little until the bouncing is done

Maybe the code at the [`develop` branch](https://github.com/ClemensGruber/hani-mandl/tree/develop) already includes those changes, and others. It is not easy to figure out because commit messages have been obfuscated.

With kind regards,
Andreas.

amotl, 2023-06-22:
> What about doing proper versioning using Git tags, and continuing development on the main branch instead, @ClemensGruber?

It would be really sweet, to save others from needing to spend hours of debugging to resolve this issue.

ClemensGruber, 2023-06-22:
`master / main `is now updated to the latest `develop `branch and rotary should also work with now work.

closed 2023-06-22 by ClemensGruber