hiveeyes/hanimandl #38: Idea: low batt warning

KarlEffinger, 2023-10-04:
I run my Hani Mandl on a Heltec board via USB power supply. Sometimes I use a power bank as power source. That's why I thought about a litte soft- and hardware addition:
- use one of the A/D pins with an appropriate voltage divisor to measure the "5V" voltage
- test for the voltage every 60 seconds and
   - warn, if the voltage droppes below 4.5 V
   - close the servo and stop operation with appropriate screen message
This would prevent running low on battery while the servo is open.

ClemensGruber, 2025-03-18:
This may make sense with a bare battery as power source. But power banks have a step-up-converter that tries to stabalize the 5 V output even if the internal LiPo discharges. So I fear we will never see voltage below 4.5 V but just a abrupt shutdown.

KarlEffinger, 2025-04-01:
Clemens, you are right!

closed 2025-04-01 by KarlEffinger