hiveeyes/hanimandl #55: Add option for choosing rotation direction in setup

rallef80, 2024-03-04:
You recently added my design suggestion with a gear arm. 
(https://github.com/hiveeyes/hanimandl-hardware/tree/main/variants/variante-5_dresden) 

In order to be able to use this design, the option "#define QUETSCHHAHN_LINKS" must always be activated in the source code. However, this will only be possible for people who are confident in programming. For this reason, the question is: Could you possibly offer compiled versions for this direction of rotation in future releases, or even better: Could you also make the direction of rotation selectable in the setup? I think that would considerably reduce the hurdles to rebuild the geararm-desgin.

amotl, 2024-03-31:
Hi Ralf,

thank you for writing in. Sure, we will be happy to follow your suggestions. We will need to consider how we can ship binary builds for variants, like you are proposing.

With kind regards,
Andreas.

ikitzmann, 2024-07-08:
@amotl This is exactly were I currently fail. I have build the Hardware and flashed the original Firmware (https://github.com/hiveeyes/hanimandl/releases/tag/v0.2.13) and everything seem to work - apart from the servo direction.

I tried building my own firmware with the
`#define QUETSCHHAHN_LINKS`
(like described here https://hanimandl.readthedocs.io/en/latest/sandbox.html). But how do I proceed with the resulting "firmware.bin" file? I understand that the release is a "combined" bin file. But how do I create such a combined bin from that firmware.bin? Or how can I flash this bin with the flash download tool?

I tried several thinks with the flash download tool but none resulted in a running HaniMandl.

For me, one of the following would be really helpful (in descending order):

1. A preference in the software for the servo rotation without the need for a specific binary
2. A binary build for the variants (QUETSCHHAHN_LINKS in my case) in the releases
3. Documentation on how to create a combined firmware with the firmware.bin as part of the sandbox.html
4. At least some documentation on how to flash the firmware.bin as part of the sandbox.html

Kind regards,
Ingo

ikitzmann, 2024-07-11:
After many tries with platformio in Windows and WSL, I was now able to successfully create and flash a build by:

1. Following the instructions at https://community.hiveeyes.org/t/hanimandl-halbautomatischer-honig-abfull-roboter/768/17#alternative-2-firmware-selber-bauen-3 to build with the Arduino IDE but sticking to the lib versions of https://community.hiveeyes.org/t/hanimandl-halbautomatischer-honig-abfull-roboter/768/17#test-with-latest-versions-9
2. Export the compiled firmware from the Arduino IDE (Ctrl + Alt +S)
3. First flashing the original combined firmware with the flash download tool
4. Use the flash download tool to flash this bin at position 10000

I did not flash this with the Arduino IDE, because I built the HaniMandl for my brother and it was already at his place. So I needed a way to create a firmware which **_he_** could flash easily.

Nevertheless: it would be much easier to have a setting in HaniMandl to reverse the servo (e.g. as part of the servo settings). This would not need a dedicated firmware.

amotl, 2024-08-24:
Dear @ikitzmann,

thanks a stack for writing in. Your requests sound sensible. We will check what we can do for you.

With kind regards,
Andreas.